Rahix/shared-bus #1: Question about the origin of the problem

TeXitoi, 2018-08-29:
I think I see the goal of this crate: device drivers take an I2C object by value to use it. But an I2C bus can have several peripherals attached. Thus you need this to share the I2C bus between the drivers.

But the real solution should no be that the driver take a `&mut I2C` on the method it need, without taking ownership of the bus? Is there some patterns that can't work like that? Is there some pattern that would not be possible with shared-bus, but possible with taking a `&mut I2C`?

Rahix, 2018-10-19:
Sorry for the late response.  Take a look at the [relevant section](https://blog.rahix.de/001-shared-bus/#not-owning-the-bus-peripheral) of my [blog post about `shared-bus`](https://blog.rahix.de/001-shared-bus/) to see why I don't think your proposal is a good solution.

TeXitoi, 2018-10-19:
> It can't keep the same strict guarantees that a device implementation that owns the bus does

That may be true on linux, but that's not true on most embedded hal implemetation as each peripheral has its own type.

> A driver developer needs to expose when the bus is needed. This makes the driver's API a lot less ergonomic and leaks implementation details.

That's not a black and white thing: knowing that this method will block the bus while the other will not can be very important.

> An application developer needs to keep track of two things instead of one. Imagine a function that wants a reference to the device. With this approach, it would need two parameters instead of one

That's the same argument as above.

> Requiring the bus as a parameter makes generic drivers a lot harder to write.

```rust
trait TemperatureSensor {
    type Bus;
    fn read_temperature(&mut self, &mut Bus) -> f32;
}
```

And it would work.

Now, I have a use case when SharedBus doesn't work.

I have 3 tasks (I'musing RTFM):
 - a very important real time task that must preempt anything, doesn't need the bus
 - a medium priority task using the bus (reading some sensor value)
 - a very slow task that use the bus (a very slow display module)

With shared bus, doing the very slow task will block the very important real time task.

OK, now, I'm OK with you: SharedBus has its use case.

But I think that drivers should use the `&mut Bus` pattern because it's trivial to implement the owning implementation on top of the `&mut` implementation, while the contrary is not possible.