Rahix/shared-bus #10: Support for platforms without atomics?

WyseNynja, 2020-07-20:
```
$ cargo check --target thumbv6m-none-eabi
    Checking shared-bus-rtic v0.2.1 (https://github.com/ryan-summers/shared-bus-rtic?branch=feature/spi-full-duplex#fad95b37)
error[E0599]: no method named `compare_exchange` found for struct `core::sync::atomic::AtomicBool` in the current scope
  --> /Users/bwstitt/.cargo/git/checkouts/shared-bus-rtic-b510fcc9e8973e1f/fad95b3/src/lib.rs:73:14
   |
73 |             .compare_exchange(false, true, Ordering::SeqCst, Ordering::SeqCst)
   |              ^^^^^^^^^^^^^^^^ method not found in `core::sync::atomic::AtomicBool`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0599`.
error: could not compile `shared-bus-rtic`.

To learn more, run the command again with --verbose.
```

I'm not sure there's any way to "fix" this. I think it might just be better to say that `thumbv6m-none-eabi` and other platforms without atomics are not supported.

I'm going to use a platform with thumbv7m-none-eabi instead of the feather_m0 that I was planning on using since atomics are going to be helpful for a few things.

Rahix, 2020-07-20:
`shared-bus` does not have any requirement on atomics whatsoever; the error you are referencing comes from `shared-bus-rtic` which is maintained by @ryan-summers.

ryan-summers, 2020-07-20:
Sorry for the unfortunate name @Rahix - the original intention of `shared-bus-rtic` was just to provide convenience wrappers around `shared-bus`, but there were some issues with the `Sync` implementation that make `shared-bus` not thread safe, so I ended up wrapping a new implementation. Would be happy to rename `shared-bus-rtic` to avoid confusion.

WyseNynja, 2020-07-20:
Oops! I meant to open this there. Too many tabs (and too many Ryan's!)

closed 2020-07-20 by WyseNynja