Rahix/shared-bus #29: method not found in `AtomicBool`

pdgilbert, 2022-01-30:
(Related to #4 and #17.)
Using `shared-bus  v0.2.0`  I have  several examples building on several device HALs  but with  `stm32f0xx`  I have a problem  with `.compare_exchange`. (Also happening with git shared-bus  #0f429435.)

``` 
Compiling shared-bus v0.2.2 (https://github.com/Rahix/shared-bus#0f429435)
error[E0599]: no method named `compare_exchange` found for struct `AtomicBool` in the current scope
   --> /home/paul/.cargo/git/checkouts/shared-bus-afeeec54cdff0f33/0f42943/src/mutex.rs:208:14
    |
208 |             .compare_exchange(
    |              ^^^^^^^^^^^^^^^^ method not found in `AtomicBool`
```
CI with examples working on other device HALs is at https://github.com/pdgilbert/rust-integration-testing/runs/4998981227?check_suite_focus=true. (`stm32f030` is a job at the bottom.)

Rahix, 2022-01-31:
That's a thumbv6 platform, right?  I think we need to use [`atomic-polyfill`](https://github.com/embassy-rs/atomic-polyfill) as a workaround...

pdgilbert, 2022-01-31:
Yes,  `thumbv6m-none-eabi`. I suppose I would have the same problem testing on `stm32l0xx-hal` but I have that disabled for other reasons.

Rahix, 2022-02-01:
Do you want to look into integrating `atomic-polyfill` or should I take care of it?

pdgilbert, 2022-02-01:
@Rahix could you please take care of it. (It is somewhat beyond my current state of understanding and I am already stretched pretty thin.)

Rahix, 2022-02-01:
No problem, I'll do it.

PR #30 (2022-02-09): Use atomic-polyfill for AtomicCheckMutex
This will allow using `shared-bus` on thumbv6 platforms which do not have real atomics.

@pdgilbert, can you please give this a try on your systems to verify that it actually works now?

Fixes #29.
closed 2022-02-09 by Rahix