Rahix/shared-bus #32: Can bus suppport?

codytrey, 2022-03-19:
Is there anything that would prevent adding support for can bus? If not and this would be a welcome addition I'll plan to start working on it as it would be useful in my own project(s).

Rahix, 2022-03-21:
Hi,

is there an existing generic trait crate for CAN bus (I vaguely remember seeing one at some point)?  If yes, it would of course be great to also support it here.  You're welcome to send a PR :)

codytrey, 2022-03-22:
That's great to hear! I've got a theoretically implementation that I plan to test tomorrow if the work day is slow. One issue I know of now is that embedded-hal-mock doesn't currently support can. If that's an issue, it doesn't look like like it should be too hard to add can support there too.

Rahix, 2022-03-22:
> If not having tests is an issue, it doesn't look like like it should be too hard to add can support there too.

Of course tests would be great but in this case I don't think we're missing too much without the tests here...

codytrey, 2022-03-23:
After some more testing it seems non-trivial to add support for can bus because while embedded-hal does have a generic trait for it, the hal crates that implement it (stm32f1xx-hal at least, as that's the only one I'm currently able to test) also depend on bxcan and the bxcan::Can trait would also need to be implemented.

There's also a more fundamental issue of how to handle can frames received for other device drivers while the bus is being used another device driver. I can think of a few ways to handle that, but are probably out of scope for this crate.

1. Implement bxcan::Can for BusProxy and allow setting the bus filter bank when mutex is locked.
    - this is probably a bad idea as it will cause frames for the other devices to be ignored (It's completely valid for a periodic status frame from device X to be received while sending frames to/receiving frames from device Y)
2. Write all received frames to a buffer, have the BusProxy read from the buffer with logic to filter out frames from other devices
    - this has a lot of over head and I could see it having scenarios that are unsound
3. Not using a shared bus, and writing can device drivers to produce and consume frames leaving the user to implement to the handling of the can interface.