Rahix/shared-bus #51: What kind of shared bus should I use?

za-songguo, 2023-08-18:
I have already wrapped my peripherals with `Arc` and `Mutex`, like this:
```
pub struct AppPeripherals<'a> {
    pub sht31: Arc<Mutex<...>>,
    pub max17048: Arc<Mutex<...>>,
}
```
(The peripherals will be shared across multiple threads.)
What kind of shared bus should I use? `BusManagerSimple` or `BusMutex`?

Rahix, 2023-08-31:
Hi, sorry for the late response.

When every peripheral lives in its own mutex, you still need one additional mutex guarding the bus itself to ensure mutually exclusive access to it.  As it seems you are using the `Mutex` from `std` for your peripherals, the [`BusManagerStd`](https://docs.rs/shared-bus/0.3.0/shared_bus/type.BusManagerStd.html) would be most appropriate.

However, you can also consider going a different route:  You could put all peripherals into a single big mutex and then ensure exclusive access using the [`BusManagerAtomicCheck`](https://docs.rs/shared-bus/0.3.0/shared_bus/type.BusManagerAtomicCheck.html).  This is particularly useful for frameworks like RTIC.  Check the linked documentation for more details.

za-songguo, 2023-09-02:
Thanks for your reply. It is useful for me.

closed 2023-09-02 by za-songguo