Rahix/shared-bus #54: Implement embedded_hal::i2c::I2c for I2C proxies

avsaase, 2023-11-15:
Would it be possible to implement `embedded_hal::i2c::I2c` for the I2C proxy types? I'm using a [driver crate](embedded_hal::i2c::I2c) for the bme280 sensor that depends on this this trait so it currently cannot be used together with `shared-bus`.

Rahix, 2023-11-16:
You mean [this one](https://docs.rs/embedded-hal/1.0.0-rc.1/embedded_hal/i2c/trait.I2c.html) from `e-h 1.0.0-rc1`?

For `e-h 1.0`, the entire I2C and SPI bus traits have been reworked such that `shared-bus` isn't really needed anymore.  Instead, you'll find simple "bus manager" implementations in [`embedded-hal-bus`](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-hal-bus) from now on.  For example, [`RefCellDevice`](https://docs.rs/embedded-hal-bus/0.1.0-rc.1/embedded_hal_bus/i2c/struct.RefCellDevice.html) would be the replacement for [`BusManagerSimple`](https://docs.rs/shared-bus/latest/shared_bus/type.BusManagerSimple.html).

avsaase, 2023-11-19:
I wasn't aware of these changes in `embedded-hal 1.0.0-rc1`. I'll close this issue because the ecosystem seems to be moving to `e-h 1.0`.

closed 2023-11-19 by avsaase

hacknus, 2024-06-21:
sorry to open this up again, but as I understand in embedd-hal-bus, sharing buses between tasks is only possible with critical sections, whereas with embedded-hal 0.2 and shared-bus, it was much easier to share the bus across different tasks. Will shared-bus be further updated to embedded-hal 1.0?

Rahix, 2024-07-14:
@hacknus, `shared-bus` also just used critical sections for cross-task sharing.  It just wasn't that visible.

hacknus, 2024-07-14:
@Rahix yes I figured.. but with embedded-hal 1.0 i'd have to implement all the critical sections myself and it's not as elegant as with shared-bus, where I was able to share the i2c bus across different tasks in freeRTOS for example. every driver was able to accept the bus without any cs wrapped around it. or am I missing something and this is also possible in embedded hal 1.0?