Rahix/shared-bus #60: bumping to embedded-hal-1.0

hacknus, 2024-10-02:
Are there plans to bump this to embedded-hal-1.0?
I have not found an easy way to share an I2C and SPI bus between different tasks in embedded-hal-1.0 but it was super easy with shared-bus and embedded-hal-0.2 (I used freeRTOS).

Would be happy to help out!

clintfred, 2024-10-02:
A PR exists to do this: https://github.com/Rahix/shared-bus/pull/55

It's been sitting un-merged for quite some time, however.

Sycrosity, 2024-10-04:
the PR is mine and the repo is: [here](https://github.com/Sycrosity/shared-bus) incase you want an alternative for the meantime. However, if you are using embassy (or are fine with having embassy parts in your project), I'd recommend using an `embassy_sync::Mutex` instead (e.g. `Mutex<NoopRawMutex, I2C<'static, esp_hal::peripherals::I2C0, Async>>`)

rursprung, 2024-11-13:
in my understanding you don't need `shared-bus` for `embedded-hal` 1 anymore, you can use [`embedded-hal-bus`](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-hal-bus) instead. `shared-bus` is only needed for `embedded-hal` 0.2.x

probably there should be a `shared-bus` 0.4.0 release which drops the e-h 1.0.0-alpha.9 dependency (& support for it) and adds a fat note to the README about `embedded-hal-bus`?

hacknus, 2024-11-13:
Then I'm probably using it wrong (freeRTOS)...

```rust
use embedded_hal_bus::i2c;
```

```rust

// initialize i2c1
    let scl = gpiob.pb6;
    let sda = gpiob.pb7;
    let i2c1 = dp.I2C1.i2c(
        (scl, sda),
        i2cMode::Standard {
            frequency: 100.kHz(),
        },
        &clocks,
    );

    let i2c_ref_cell = RefCell::new(i2c1);
    
```

```rust

Task::new()
        .name("IR SENSOR TASK")
        .stack_size(1024)
        .priority(TaskPriority(3))
        .start(move || {
            let addr = SlaveAddr::default();
            let mut ir_sensor =
                Mlx9061x::new_mlx90614(i2c::RefCellDevice::new(&i2c_ref_cell), addr, 5).unwrap();
            let mut raw_channel1;
            let mut raw_channel2;
            let mut ambient_temperature;
            loop {
                raw_channel1 = ir_sensor.raw_ir_channel1();
                raw_channel2 = ir_sensor.raw_ir_channel2();
                ambient_temperature = ir_sensor.ambient_temperature();
    
                CurrentTask::delay(Duration::ms(100));
            }
        })
        .unwrap();

 Task::new()
        .name("IR TEMPERATURE TASK")
        .stack_size(1024)
        .priority(TaskPriority(3))
        .start(move || {
            let mut raw = TemperatureRawValue::Err(Ads122Error::Timeout);
            let mut temperature = TemperatureValue::Err(Ads122Error::Timeout);

            let i2c_addr = 0x40;
            let sensor_type = ADS122SensorType::PT1000;
            let mut temperature_sensor =
                TemperatureSensor::new(i2c_addr, i2c::RefCellDevice::new(&i2c_ref_cell));
            let mut temperature_sensor_state = temperature_sensor.init(sensor_type).is_ok();
            loop {
                 // do stuff
            }
        })
        .unwrap();

```

this gives me this error:

```

error[E0382]: use of moved value: `i2c_ref_cell`
   --> src/main.rs:229:16
    |
170 |     let i2c_ref_cell = RefCell::new(i2c1);
    |         ------------ move occurs because `i2c_ref_cell` has type `RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>`, which does not implement the `Copy` trait
...
208 |         .start(move || {
    |                ------- value moved into closure here
...
211 |                 Mlx9061x::new_mlx90614(i2c::RefCellDevice::new(&i2c_ref_cell), addr, 5).unwrap();
    |                                                                 ------------ variable moved due to use in closure
...
229 |         .start(move || {
    |                ^^^^^^^ value used here after move
...
236 |                 TemperatureSensor::new(i2c_addr, i2c::RefCellDevice::new(&i2c_ref_cell));
    |                                                                           ------------ use occurs due to use in closure

```

rursprung, 2024-11-13:
i don't see you using `embedded-hal-bus` in your code?
but i'm no expert in this topic and haven't used freeRTOS. maybe ask in the [rust-embedded matrix room](https://matrix.to/#/#rust-embedded:matrix.org), someone there might know?

hacknus, 2024-11-13:
@rursprung I'm creating a RefCellDevice in the constructor calls for the Devices:

```rust
i2c::RefCellDevice::new(&i2c_ref_cell)
```